VerifiableRobotics/LTLMoP #93: Regarding specEditor.py

Shradha131, 2017-11-08:
The tool is installed in the same way as written in the installation guide but whenever I run specEditor.py, a GUI window is popped up and it does not allow me to save my project. It says, there is no attribute name SAVE ( screenshot is attached below). It would be a great help if any1 can guide me on this.

[image]

mattdanhall, 2017-11-17:
I am having the same issue, have you managed to resolve it?

layashamgah, 2017-11-17:
I guess this is due to updated version of wx. I still don’t know how to solve the problem

Shradha131, 2017-11-17:
Yes, it was because of updated version of wx.
 We tried to resolve the issue by finding the replacement of each old
function of wx by the new one but we got stuck in one function named
beginDrawing which is not present in wx.DC so we left LTLMoP. If you find
anything on this, please let us know.

Thank you.

mattdanhall, 2017-11-20:
I have managed to get it working so far by using wxPython version 2.8

The files for install are found here:

https://sourceforge.net/projects/wxpython/files/wxPython/2.8.12.1/

I hope this helps!

Shushman, 2018-11-11:
Hi @mattdanhall , are you on Windows, Ubuntu or MacOS? None of the .dmg files for MacOS  for wxPython 2.8.12.1 (or others for that matter) seem to work